benkrasnow/DSKY_EL_replica #2: BOM?

TAGood827, 2019-05-19:
Do you have a BOM for the PCBs?

benkrasnow, 2019-06-01:
Added BOM from Kicad default.  The file could use Digikey part numbers.  Want to add?

closed 2019-06-01 by benkrasnow

TAGood827, 2019-06-01:
Thanks, Ben! I had begun compiling a BOM which includes all the parts for both PCBs and P/Ns for Mouser. I had pulled that info from KiCAD but as it does not completely define the parts, I am only certain of a few components because only a limited number returned a specific enough part I was confident was correct. I have taken my best guess on them. I would ask you to review and if you find errors to please advise with more detail so I can correct and return to you.

I would also like to ask if it might be possible to contract you to manufacture just a few of the glass plates for me? I purchased a set of 10 boards from PCBGOGO to just do a few units for myself and a few close friends and I have purchased enough glass for 10 pieces. I would rather not invest in all the screen printing materials if I can avoid it as I am not looking to go into production. If you are local to the Philadelphia PA area, I would be open to meeting with you to discuss further. 

This is a tremendous effort on your part and I cannot thank you enough for your extreme generosity in sharing this information with the Apollo DSKY fan community.

[Driver & Segment PCB BOMs.xlsx](https://github.com/benkrasnow/DSKY_EL_replica/files/3244254/Driver.Segment.PCB.BOMs.xlsx)

benkrasnow, 2019-06-02:
I added Digikey order lists (in .jpg) to the repo.  If you'd like to combine the part numbers, and add Mouser part numbers into the BOM, please submit it, and I'll be thankful and happy to add it.

I've been working non-stop on this project for a while, and want a rest to work on other things.  I don't have time to make 10 glass panels, but if there is enough interest, it may make sense to contract some work out, and maybe do some work in my shop.  I'll let you know.

TAGood827, 2019-06-02:
Thank you Ben! I'll update the BOM this afternoon and get it back to you.

TAGood827, 2019-06-02:
Okay, here's the integrated BOM with both Mouser and DigiKey P/Ns. I made a couple changes to the parts I had selected so there was full equivalency between Mouser and DigiKey. I had to change one of the caps as Digikey had no stock and a 54 week lead time. They listed several other equivalents and I chose the one that had a Mouser equivalent (Mouser doesn't stock the Samsung caps). Of the other items that you did not have an image for (those items in the list without a V in the Verified column) , I just crossed my choices from Mouser to find a DigiKey equivalent. Again I had to change a couple here and there but otherwise most of hem crossed directly.

There were a couple items shown in your images that I wasn't sure about. I could not match up their descriptions with anything in the KiCAD files. They are listed at the bottom of the spreadsheet.

I changed my ARM micro choice for just the chip since I was definitely wrong on that. There was an ATMEL ATMEGA programming tool that might have worked to program the chip onbard but It doesn't look like it is available anymore.

https://www.amazon.com/HobbyKing-Atmega-Socket-Firmware-Flashing/dp/B00V2W467I/

If I need to correct anything else please advise.

Thanks again for your assistance with this!

[Driver.Segment.PCB.BOMs.xlsx](https://github.com/benkrasnow/DSKY_EL_replica/files/3244983/Driver.Segment.PCB.BOMs.xlsx)

TAGood827, 2019-06-03:
Got my boards today, but it looks like they did not listen when I asked them to plate and mask on the bottom side of the display board only.

Do you think it is possible to remove the mask safely from the one side?

[image]

benkrasnow, 2019-06-04:
Probably the best way to salvage the board is to attach some 600 grit sandpaper with double-stick tape to a piece of glass, and then sand the solder mask down by hand until the segments are all visible.  This will probably work pretty well, and will leave you with a super flat surface for attaching to the painted glass display.  It's difficult to tell from the photos, but it looks like there might be solder on the exposed pads (HASL process?)  You'll pretty much always want to choose electroless nickel immersion gold (ENIG).  It makes soldering surface-mount parts much easier.  If you order more boards without mask, definitely choose ENIG for the segment boards.

The pushbutton, LEDs and their current-limiting resistors, and even caps can be lifted from the Adafruit Trinket board as well.

The unused parts on the Digikey were for other projects.

TAGood827, 2019-06-25:
Hello Ben,

I saw that you had your display hooked up to the AGC and the DSKY with Carl
Clauch! Congrats on that! Does that require different code to be loaded on
your display?

I also wanted to check in with you again about the EL DSKY glass. I have 9
other guys organized to build one of your displays using the boards I had
made. Before I order the parts from the BOM I worked out with you, I just
wanted to see where you stood regarding my request for you to contract
making the glass for me. I know you desired to take a break from the
project which I fully understand and again, I'm not asking you to do them
now. I would just like to know if you are willing to do them in the near
future. I would also like to ask if you could provide an estimate for your
services with me providing the glass I have to you. I would like to provide
this estimate to the rest of the guys to see if it would put them off doing
the build.

Thanks again for your efforts to produce this display and for sharing it
with all of us.

Tim

cjkeithley, 2019-07-03:
Hi Ben,

I was watching your video and noticed that you mentioned that you'd added some current limiting resistors to the HV509 outputs that drive common. But, I don't see these resistors on the driver or segment schematics... am I missing something or looking in the wrong place?  

> I've been working non-stop on this project for a while, and want a 
> rest to work on other things. I don't have time to make 10 glass panels, but 
> if there is enough interest, it may make sense to contract some work out, and 
> maybe do some work in my shop. I'll let you know.

I'm interested as well!  I'd buy 2 or 3.

Thanks & best regards,

Craig

cjkeithley, 2019-07-03:
Make that HV507 (not HV509)

benkrasnow, 2019-07-04:
Tim, the firmware is the same.  Carl wanted to send data via physical serial port (not over USB), and luckily the SAMD21 can accommodate this for 19200 baud, which is fast enough to load the display at a fast frame rate.

I cannot commit to making a batch of glass panels, unfortunately.  Have you tried asking a traditional screen-print shop?  If you give them the artwork, and paints, they could likely do the job at higher quality and much faster than I could.

TAGood827, 2019-07-04:
Okay Ben, thanks for the info.

I haven't looked into any local shops yet. I'll have to start working on
that.

Thanks again!

Tim

TAGood827, 2019-07-04:
One other question, regarding the phosphor color. You had said it was a
little too blue. Is there a way to adjust that?

Tim

benkrasnow, 2019-07-07:
Lowering the drive frequency will shift the phosphor color to be more green, and less blue.  It will also make the display dimmer, so it seems 750Hz or thereabouts is a good compromise.  In earlier EL projects, I was driving at 4KHz.  I sent a note to Lumilor asking about greener colors, and they responded quickly and helpfully, but didn't have a way to supply any other colors.  I also briefly researched making a phosphor, and decided that was too much work.  I also haven't seen any color-corrected photos of a working DSKY, and have never seen one working myself, so it's hard to say exactly what it looked like despite the "530nm" specification on the official documents, which is indeed pretty deep green.