benkrasnow/DSKY_EL_replica #5: Airbrushing the Lumilor paint

cjkeithley, 2019-07-07:
Hi Ben,

Since mid last year I've been contemplating using the Lumilor paint to create a DSKY EL digital indicator, so I was pleased to see your project was a success.  I hesitated because Lumilor says you need to use an HVLP gun with their paint.

I don't know enough about airbrushing to have an opinion on using an HVLP gun, but I was startled to discover that most HVLP guns require 5+ CFM, and that even my middling air compressor puts out less than 2CFM... and that compressors that can put out 5+ CFM can be expensive.  Sadly, Amazon doesn't specify the CFM required by the PowRyte guns.

Where am I'm going with this?  

A question:  did you use a high (5+) CFM compressor with that small PowRyte HVLP gun, or did you use a standard work shop (or air brush) compressor (<2 CFM)?  

If you used a <2 CFM compressor with the small PowRyte gun then my existing compressor should suffice as well.  If needed to use a high CFM compressor with the PowRyte then I'll likely trying using a Paasche HG-08, which (best case) only requires 1.5 CFM.

Thanks,

Craig

benkrasnow, 2019-07-07:
I've used https://www.amazon.com/gp/product/B004INERK4/ and https://www.amazon.com/gp/product/B01G61DNF0/ to spray Lumilor with an average sized 110V shop air compressor (not sure of CFM, but manufacturers all exaggerate so much, it doesn't matter).  It works fine since you'll only need high air flow from the tank for short bursts while painting, and the compressor can keep up in between.  Unless you are painting a whole car, the compressor tank will be able to supply as much air as you need for small objects.

cjkeithley, 2019-07-07:
great advice, thanks!

closed 2019-07-07 by cjkeithley