benkrasnow/DSKY_EL_replica #6: VBUS signal line on X1 USB connector

rrainey, 2020-07-06:
Ben, do you ever use the USB interface on the Driver board? I'm noticing that VBUS (pin1) on X1 doesn't seem to be tied to anything. That's a bit different than the Trinket M0 that basic design seems to be derived from. 

I'm guessing that means that you'd first power up the board using the power plug before connecting the USB. Just curious.

benkrasnow, 2020-07-06:
The design requires more power (and voltage, if I recall) than USB can deliver.  So, I made the choice to only power the device from the external supply.  You could power up the MCU via USB, but then nothing else on the device would work, so that doesn't make sense from a usability standpoint.

rrainey, 2020-07-06:
Thanks, that all makes sense. Does that mean you are not using the USB for any purpose at all, or simply that you are taking power from the power plug but will still occasionally hook up the USB for talking to the MCU?

benkrasnow, 2020-07-06:
USB is how the MCU is programmed, and the only way that data are sent to the display in operation, so yes, it is used.

rrainey, 2020-07-06:
I'm asking as I look ahead to the development/test/debug required to integrate a display board into a larger DSKY replica of my own design.  If I need to resort to SWD/SWC prgramming, I can -- just wondering what to expect might work ...

rrainey, 2020-07-06:
Got it. Thanks

closed 2020-07-06 by rrainey