benkrasnow/DSKY_EL_replica #7: HV507 Common line: diode protection and current limiting

rrainey, 2020-07-06:
Hi Ben,

I was looking at the last committed Segment PCB layout in GitHub and comparing that with your YouTube walk-through video. 

I see what looks like the Common protection diodes you mentioned at the end of your video already on the Segment PCB (1)-- I see sixteen parallel Common lines rather than the eight you mentioned in the video, though (2).

I'm guessing you made a late change in the number of Common driver lines subsequent to producing the walk through video -- correct?

I'm guessing we'd still put one current limiting resistor on each Common driver line (you mentioned a value of 47K Ohms elsewhere) -- if all that is correct, I might just put together an updated board layout and get a fresh batch of Segment boards made.

Your thoughts?

[image]

benkrasnow, 2020-07-06:
If I were going to redesign it, I'd add a current-limiting resistor to each line immediately at the IC's output.  Then, downstream of the current limiting resistors, 8 or 16 lines could be joined together for better drive strength.  I would not build it as shown in #2, since even if all lines are switched exactly at the same time, mismatch in output transistors causes problems (ie the IC is destroyed).  The clamp diode is good insurance, so I'd keep all that as-is.

rrainey, 2020-07-06:
Well, I'm not sure I have the stomach that scale of a redesign at this point -- the driver PCB is really packed (at least for my skill level). Some substantial layout changes would be required to add the resistors next to the chip as you suggest. Am I quoting the right value for the resistors - 47K? 
I was going to use 0.2W resistors (the HV507 data sheet rates the output of each line at 0.5mA). 
Thanks for your support!

benkrasnow, 2020-07-06:
You can add the resistors to the segment board -- same step as placing and reflowing the board-to-board connectors.  I think you'll definitely want to add resistors to at least the 8 or 16 common lines -- don't connect them directly together as shown.  It may work for a while, but it's probably not a good plan.  Ideally, the resistors would be added to every segment, but it's likely that it will work fine only putting resistors on the commons only.  The resistors will dissipate very little average current, so 0402 or even 0201 would be fine.  Don't worry about the power rating.  47K minimum.  Maybe even go 100K.  I'm sorry for the lack of precision in my design -- it's not a problem of missing documentation, but rather a lack of having a completely final ready-for-production design.  There are still some small tweaks needed.

rrainey, 2020-07-06:
Thanks! will do.

rrainey, 2020-07-07:
I played with adding resistors to the Segment board in KiCad (image below). Having overnight to ruminate about this, though, I think I've decided to try placing the limiting resistors on the driver board as you suggested. The design already requires components on both sides of the board -- having the resistors on the connector side doesn't fundamentally complicate anything.

I'm having a few issues bringing up these projects for editing in KiCad. As is, it was complete enough that I could send the KiCad files to CircuitHub for production. I might need some pointers getting the Driver board project editable. I'll keep you posted.

[image]

rrainey, 2020-07-07:
I toyed with adding resistors to the Segment board last night (image below).

I'm going to think more about your suggestion to place them on the Driver board instead. I'll try to commit to a choice this week.

[image]

closed 2020-07-07 by rrainey