benkrasnow/DSKY_EL_replica #8: Current limiting resistor array modification

rrainey, 2020-07-07:
I worked this change up last night.  Sixteen 0402 75K (or 100K) resistors. 
CircuitHub will make a batch of 10 for about $220 and 20 for only $20 more.  Want any?

[image]
[image]

I can send you a pull request with the modifications.

benkrasnow, 2020-07-12:
Sure send a pull request for the resistors.  I haven't worked on the DSKY in quite a while, so don't have a need for PCBs.  I think you'll find a lot of folks that want to buy them, though!

rrainey, 2020-07-13:
Great. I just submitted the pull request. I'll close this Issue.

commit 64473083 (2020-07-15): Merge pull request #9 from rrainey/master

Add 0402 size current limiting resistors. Update PCB version markings; Closes #8
closed 2020-07-15 by benkrasnow