benkrasnow/DSKY_EL_replica #12: Not an issue; Just an interesting reference on EL current limit resistor selection.

rrainey, 2020-12-08:
In doing some research on selecting an "optimal" value for current limiting resistors in EL Display Driver design, I came across this reference: https://patents.google.com/patent/US5095248

I suspect this might be applicable here -- any thoughts?

benkrasnow, 2020-12-10:
I gave it a quick read, and don't immediately see anything applicable.  Can you be more specific?

rrainey, 2020-12-10:
There was a relation buried in there regarding a guideline for selection of current limiting resistor values. I realize the overall design of their driver is a bit different, but I was wondering if it might be applicable.  I know we talked about using 100k or 75k resistors, but I was just wondering if there was a more deliberate way to make a selection.

You'd have to examine the patent document to get the meanings of variables, but this was the relation to be enforced:

Va - (VD (on) + ID (on) × Ri) ≧ 2 (VTEL + VMOD)

benkrasnow, 2020-12-11:
It looks like the patent is describing an active matrix driver circuit for EL display elements.  The design constraints are different since the bipolar AC waveform is being supplied from an external source, and the resistor is used to prevent the tiny pixel-level driver from passing too much current when the pixel is illuminated constantly.  The HV507 circuit in this project starts with a unipolar high voltage supply and fakes this into a bipolar supply by switching both sides of the EL element, which is where the trouble begins .  Switching both sides of the element (as opposed to just switching on/off an existing AC waveform) creates a much bigger possibility for overshoot, which is what happens when the clamp diodes in the HV507 get overwhelmed.  The current limiting resistors protect the clamp diodes, which then protect the rest of the IC.  It's not a constant power dissipation problem.

rrainey, 2020-12-12:
Thanks. Got it.  BY the way, I was looking at alternatives for adding all the requisite resistors.  One goal -- for me at least -- is to avoid making huge changes to your original implementation. Arrays seem relatively expensive and also take up quite a lot of PCB real estate (at least when you get 14 or so 12-way arrays placed). I even considered adding a third board to sandwich in between the Segment and Driver boards where the resistors would be placed -- that has the virtue of being insert-able into the assembly without altering the original two boards.  I have not completely abandoned that idea, but for now I am just adding the 161 new discrete resistors to the Segment board.  

I started by cataloging the three connector connection list (attached)
[Display J pinouts - Sheet1 - Display J pinouts - Sheet1.pdf](https://github.com/benkrasnow/DSKY_EL_replica/files/5682160/Display.J.pinouts.-.Sheet1.-.Display.J.pinouts.-.Sheet1.pdf)

closed 2020-12-13 by rrainey